tinyvision-ai-inc/pico2-ice #2: SWD connector is vertical, can this cause issues in mailing?

vr2045, 2024-12-02:
The vertical connector for SWD was added to reduce congestion: should this be horizontal to minimize the possibility of breakage during shipment?

timoksn, 2024-12-03:
if it's placed horizontally, would it be placed on the opposite side of the USB connector (by SW2) ?

vr2045, 2025-01-26:
SWD connector was originally flat but due to board congestion, we had to make it vertical.
Options:
1. Keep it vertical. If we do see this connector getting ripped off during mailing, we can consider changing it or changing the packaging so its more robust?
2. Change to horizontal: lack of board space. 

For now, we will try to make this change in the next rev of the board, lets see...