tinyvision-ai-inc/pico2-ice #4: Suggestion: Add silkscreen PMOD id number

timoksn, 2024-12-02:
Add silkscreen PMOD id number if there's enough space
This could help with documentation and tutorials

[image]

vr2045, 2025-01-26:
The issue shows up with the 2 PMOD's on the left side of the board. 
I have called the lower left PMOD as "iCE PMOD A" and the one on the top right "iCE PMOD B". The ones on the right side are already called "RP PMOD" and "RP-iCE PMOD" which should be sufficient.

vr2045, 2025-01-27:
[image]

[image]

closed 2025-01-27 by vr2045