tinyvision-ai-inc/pico2-ice #6: Power monitor cannot measure voltage of a battery connected to Vin...

alastairpatrick, 2024-12-05:
...because D5 prevents current from that source from flowing through the power monitor potential divider. A simple fix would be to connect the power monitor potential divider to Vbus instead of Vin. However, in some applications, this makes things worse, because the battery drains faster on account of the current through the potential divider. Idea 1: use a voltage follower, either BJT transistor or op-amp based, to reduce current through the potential divider. Idea 2: allow power monitoring to be turned on and off based on level of ADC6, so that no current flows when power monitoring is not in progress.

alastairpatrick, 2024-12-05:
Idea 3: or just remove the power monitor potential divider altogether. ADC7 is on a pin, so it would be easy for a user to connect a potential divider to it, with whatever approach to avoiding wasteful energy drain, externally.

alastairpatrick, 2024-12-05:
A related issue, so I'll put it here. When power is connected to Vbus (or Vin if the power monitor potential divider was moved to Vin), ADC7 will rise to half of the supply voltage. Some time later, delayed by soft start, the 3.3V regulator will bring RP IOVDD up to 3.3V. In the interim, ADC7 is more than 0.5V above IOVDD. ADC7 is not a fault tolerant IO, so it's outside briefly absolute maximum rating. It could be fixed with a clamping diode across R29.

alastairpatrick, 2024-12-05:
I like this as a compromise. By default, there is no wasteful battery drain. User can use the power monitor by adding one external resistor. They can tailor the resistor value to the battery voltage.
[image]

alastairpatrick, 2024-12-05:
Adding R29 makes it possible to connect directly to the positive terminal of a nominally 3.6V lithium battery. R29 is 2.2K so that at fully charged battery at about 4V yields 3.3V on GPIO47_ADC7. An external resistor in series with R29 allows for higher battery voltages. R29 acts as a current limiting resistor if ADC7 is misused, e.g. connected directly to the positive terminal of a 12V battery
[image]
.

vr2045, 2025-01-27:
This makes sense.
Implemented on the board. Thank you for the feedback!

closed 2025-01-27 by vr2045