tinyvision-ai-inc/pico2-ice #7: Switch bottom contact FPC with top contact FPC...

alastairpatrick, 2024-12-05:
... to be consistent with the Adafruit RP2350 Feather board. This could help reduce user error, such as connecting the pico2-ice to a board in a way that causes reverse polarity or a short. This would require some routing changes. In order to preserve the currently good routing of the HSTX differential pairs, I suggest leveraging the flexibility of HSTX bit crossbar to reverse the pin assignments of the HSTX pins. The assignments of GPIO10 & 11, SCL and SDA need not, I think, be assigned to different RP pins, but would have to be routed differently. This will require some changes to software written for the Adafruit RP2350 Feather board to make it work on pico2-ice, but it should be minor.

timoksn, 2024-12-05:
two possible options
-----
-----
**Slide Lock (~0.7USD) -  Mfr.#: F32Q-1A7H1-11022** 
[image]
https://www.mouser.com/ProductDetail/Amphenol-Aorora/F32Q-1A7H1-11022?qs=xZ%2FP%252Ba9zWqbp%2FwxqDihMFA%3D%3D

-----
**Flip Lock (~$1.7USD) -  Mfr. #:FH12A-22S-0.5SH(55)** 
[image]
https://www.mouser.com/ProductDetail/Hirose-Connector/FH12A-22S-0.5SH55?qs=wXhBCuSapIMPTS3Fn6eKNQ%3D%3D

vr2045, 2025-01-26:
I do like the flip lock. Will use that for the next version of the board.

I like the ability to keep the FW the same as what the Adafruit has. Would prefer to not make changes as a result.

Regardless, we should document the right way to insert the flex clearly. Especially if its different than the Adafruit. Worst case, if theres a short, this should not damage anything. We need to analyze this a bit.

vr2045, 2025-01-26:
If the flex is flipped, +3V3 gets shorted to ground. The PTC fuse will limit current and dissipation through this board.
Its not clear what happens at the other end though.

vr2045, 2025-01-27:
Resolved by using the same connector as on the RPi reference design: SFV22R-2STE1HLF

[image]

closed 2025-01-27 by vr2045