tinyvision-ai-inc/pico2-ice #12: Revisions to the Kicad Files

frohro, 2025-07-23:
Hi All,

I am teaching Digital Design this fall quarter, and am planning to make an order of pico2-ice units to JLCPCB.   To get ready for this, in the design files, I have added:

- A bunch of datasheets that were missing.  (I did not include resistor or capacitor datasheets.)
- LCSC Part numbers
- 3D Models (I'm working on this now.)

I saw the note about the SWD connector maybe causing shipping issues.  For me, this isn't really a problem as I am ordering ten shipped here to be used here, not reshipped.  Are there other issues I should address?   Would you like me to do a pull request (after the class is over to make sure bugs have been discovered)?   I have tested the pico-ice  to make sure I am happy with it, but I am planning to move on to the pico2-ice (which I have not played with), because it is the one that is receiving the love now, and is better, and the price to order from JLCPCB is about the same as if I made an order for the pico-ice.  Are there any other bugs you are aware of that I should think about addressing?  I would also be happy to share my changes with you earlier, especially if you wish to make some changes too, or would like to add some boards to my order of ten, for yourselves.

Thanks for all your good work on the pico-ice series!

Rob

frohro, 2025-07-24:
Did you have any problems with the fact your layout is different than what was recommended in the [RP2350 datasheet page 452](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf)?  I substituted a [different inductor](https://jlcpcb.com/partdetail/AbraconLLC-AOTA_B201610S3R3_101T/C42411119), because I could not find the one you specified.

josuah, 2025-08-11:
I think most of the activity happens on the chat server, you will have better luck of getting an answer there.

> There was also a problem with long start time for the crystal, which might have been because the layout is a bit different to the recommendation?

> Yes, that's been fixed on the purple version...
> We followed the reference design AFAIK...

https://discord.gg/JyHuNKHXbC

vr2045, 2025-08-11:
No changes planned as of now. Users haven't reported any significant issues so that far.
Good luck with the course!

frohro, 2025-08-12:
Thank @vr2045 and congratulations on that!  Even just adding JLCPCB part numbers, 3D models, etc. makes me a little nervous to order as many as I need.  :-)  I plan on it though.
Thanks again!
Rob

vr2045, 2025-08-12:
Good luck on this! Do share your experience with the course and we can propagate it to other profs who are using this board in their courses as well.

Do watch for the oscillator load cap change to 10pF. I think you saw this already on the Discord...

Please feel free to make a PR to the repo with your changes so we can include this in the design for other users as well!
If you do decide to submit a PR for this, please dont include any schematic or layout changes that affect the gerber itself.

closed 2025-08-12 by vr2045